wollewald/MPU9250_WE #1: MPU9250 Magnetometer does not respond

Andyinater, 2021-05-21:
Sorry if this is inappropriate, but your sketches work perfectly for me besides the magnetometer. I could not see anything on your blog for troubleshooting this. Do you perhaps know what can cause this? I am using this breakout board with my ESP32:  https://www.amazon.ca/CANADUINO-InvenSense-Accelerometer-Magnetometer-Compatible/dp/B07M76H9CM

Any help is appreciated, danke.

wollewald, 2021-05-21:
Hi @Andyinater, can you tell me more about the issue? Does is not compile, do you get wrong result or no results? That will help me to resolve it.

wollewald, 2021-05-21:
@Andyinater , I have just tried it with an ESP32 and it worked. My MPU9250 looks like yours apart from the label which is MPU9250/6500 on mine and MPU92/65 on yours.  But that shouldn't make any difference. 

I tried the unchanged MPU_9250_all_data.ino and MPU9250_magnetometer_data.ino sketches. Which one did you try? And did you change anything? 

Do you get the message "Magnetometer does not respond"?

Andyinater, 2021-05-21:
Hi @wollewald, it compiles fine. When running the magnetometer example sketch "MPU9250_magnetometer_data.ino" 

`  if(!myMPU9250.initMagnetometer()){
    Serial.println("Magnetometer does not respond");
  }
  else{
    Serial.println("Magnetometer is connected");
  }`

Results in the serial print "Magnetometer does not respond". Could it be because I am using custom pins for I2C? Is there a global variable I can define I2C pins? Sorry, I am very new to all of this.

wollewald, 2021-05-21:
If the accelerometer and the gyroscope are working then in theory there shouldn't be an issue with your I2C settings. 

If you take another sketch which also uses the accelerometer, e.g. MPU9250_all_data, do you only get the "Magnetometer does not response" message or also "MPU9250 does not respond"?

Andyinater, 2021-05-21:
Only the 'Magnetometer does not respond' message, and then it proceeds to output reasonable data for accel and gyroscope, but magneto returns all 0s

wollewald, 2021-05-21:
Then this might be related with the way I implemented the magnetometer. Maybe this does not work with every module. I have ordered a module which look exactly like yours. But this will take some time until they arrive.
Maybe I will rewrite the library and implement an alternative way of implementing the magnetometer. But I can't promise a timeline.

wollewald, 2021-05-22:
@Andyinater , I have fundamentally changed the way to retrieve magnetometer data. I have released a new version. I hope you are still motivated to try. That would help me a lot since I can't reproduce the issue with my module. 
Occasionally, but only after upload of sketches, you might need to disconnect from power and repower again.

Andyinater, 2021-05-22:
@wollewald Wow, I did not expect such a response and I greatly appreciate it. 

Unfortunately, I am still getting the same response. MPU9250 is connected, Magnetometer does not respond.

wollewald, 2021-05-22:
OK - that's of course disappointing. Maybe I can reproduce and then solve the issue when I get the MPU9250 modules that I have ordered and which look more like yours. But they come from the US and that will take few weeks. 

Have you tried another library and did it work? This one worked for me:
https://github.com/bolderflight/mpu9250-arduino

Andyinater, 2021-05-22:
For that one, I am uncertain on how to manually declare my I2C SCL and SDA pins. I attempted modifying his "mpu9250.cpp" under the method Mpu9250::Begin() by changing

`i2c_->begin();`

to

`i2c_->begin(33,32);`

Although this results in "IMU initialization unsuccessful"

Andyinater, 2021-05-22:
Honestly, at this point I'm wondering if I somehow got a knock-off unit without the magnetometer in it. If every library gives me a problem, I am the problem... lol

wollewald, 2021-05-22:
To be honest I was also not sure how to do that. There is no way to pass the SDA / SCL pins. I just took the Basic I2C example and it worked with the pins 18/19 like this:

````
#include "MPU9250.h"
TwoWire myWire = TwoWire(0);

// an MPU9250 object with the MPU-9250 sensor on I2C bus 0 with address 0x68
MPU9250 IMU(myWire,0x68);
int status;

void setup() {
  // serial to display data
  Serial.begin(115200);
  while(!Serial) {}
  myWire.begin(18,19);
  // start communication with IMU 
  status = IMU.begin();
.......

````

Good luck!

Andyinater, 2021-05-23:
So, I used the library by Kris Weiner and some new info came out by probing what is called the "Who am I" register. 

https://github.com/kriswiner/MPU9250/issues/458

It appears that I DO NOT have a genuine 9250, or I at least have a mislabeled (probably on purpose) 9255 or something. When I followed his advice in that issue, his library did end up producing magnetometer values, but I do not think I will be able to make use of them anyways.  (for my absolute orientation desires I am now looking towards the BNO055 from Bosche)

For now, I will be continuing with your library as I find it very intuitive to use. I do have a new question though: Is there a way to retrieve the offset values generated by autoOffsets()? Mainly so I can store them in flash memory so if my device accidentally power cycles I can return to logging on the previously established offsets.

wollewald, 2021-05-23:
Using the magentometer values is a bit tricky. Every cable on a breadboard influcences the measured values. In theory you could calibrate it. Just turn around in all directs and record the min/max values. Quite some effort. And you always have to calibrate whenever you change the location. I was a bit disappointed after the first experience. 

Ihave not (yet) written a function like "getAutoOffsets". But you can determine them easily yourself. It's nothing sohisticated:

````
    accOffsetVal.x = 0.0;
    accOffsetVal.y = 0.0;
    accOffsetVal.z = 0.0;
    
    enableGyrDLPF();
    setGyrDLPF(MPU9250_DLPF_6);  // lowest noise
    setGyrRange(MPU9250_GYRO_RANGE_250); // highest resolution
    setAccRange(MPU9250_ACC_RANGE_2G);
    enableAccDLPF(true);
    setAccDLPF(MPU9250_DLPF_6);
    delay(100);
    
    for(int i=0; i<50; i++){
        getAccRawValues();
        accOffsetVal.x += accRawVal.x;
        accOffsetVal.y += accRawVal.y;
        accOffsetVal.z += accRawVal.z;
        delay(1);
    }
    
    accOffsetVal.x /= 50;
    accOffsetVal.y /= 50;
    accOffsetVal.z /= 50;
    accOffsetVal.z -= 16384.0;
    
    for(int i=0; i<50; i++){
        getGyrRawValues();
        gyrOffsetVal.x += gyrRawVal.x;
        gyrOffsetVal.y += gyrRawVal.y;
        gyrOffsetVal.z += gyrRawVal.z;
        delay(1);
    }
    
    gyrOffsetVal.x /= 50;
    gyrOffsetVal.y /= 50;
    gyrOffsetVal.z /= 50;
````
For long term storage I would recommend the other procedure I have implemented with the min / max values for the axes. It's bit more effort but a one off.

Andyinater, 2021-05-23:
Thank you so much for your help, I think I will finally be able to complete my prototype.

ivicajan, 2021-09-26:
HI Wolle,
thanks for putting all that into nice library. 

I have similar problem with different pre-defined SDA/SCL that I want to pass during the setup() phase.
Is it possible to add function .setWire similar like:  https://github.com/asukiaaa/MPU9250_asukiaaa/blob/master/src/MPU9250_asukiaaa.h
lines 39-41

That way I could easily do something like this in dhe setup:
#define SDA_PIN 21
#define SCL_PIN 22
Wire.begin(SDA_PIN, SCL_PIN);
myMPU9250.setWire(&Wire);

Otherwise I can do complete init inside setup:
void setup() {
  Serial.begin(115200);
#ifdef _ESP32_HAL_I2C_H_ // For ESP32
  Wire.begin(SDA_PIN, SCL_PIN); // SDA, SCL
#else
  Wire.begin();
#endif
  MPU9250_WE myMPU9250 = MPU9250_WE(&Wire);
  if(!myMPU9250.init()){
    Serial.println("MPU9250 does not respond");
  }
  else{
    Serial.println("MPU9250 is connected");
  }

But then inside loop() it is not defined  myMPU9250 and can't compile.

Thanks!

wollewald, 2021-09-26:
@ivicajan , you can predefine the SDA and SCL pins and pass it to your MPU9250 object. I have tried this on an ESP32. 

Like this, e.g.:

````
#define I2C_FREQ 400000
#define SDA_PIN 21
#define SCL_PIN 22
#define MPU9250_ADDR
TwoWire wire1 = TwoWire(1);

MPU9250_WE myMPU9250 = MPU9250_WE(&wire1, MPU9250_ADDR);

void setup() {
  wire1.begin(SDA_PIN, SCL_PIN, I2C_FREQ);
.......
````

ivicajan, 2021-09-26:
Thanks!
it's working.

Cheers,
Ivica

wollewald, 2021-10-16:
@sparkplug23, have you tried other libraries and do you get magnetometer results with these?

sparkplug23, 2021-10-16:
> @sparkplug23, have you tried other libraries and do you get magnetometer results with these? 

I have tried others, but the values I am getting out are completely random. It doesn't even appear to be noise, it can go from +2000 to -2000 etc on each new sample. 

I dug into your library, and the function that reads the register is getting 0 for the value. _wire->available () is correctly showing as one byte, but that result (_wire->read()) is always 0. 

For now, after wasting the last 2 days I've switched to another IMU I have (gy89) and am getting sensible values with it. I'll need to spend more time trying to debug the issue with the mpu9250. I briefly looked at its datasheet (though I suspect it was an abrievated version) and didn't really see the register/i2c handshake I needed to do to get the values out of the magnetometer.

wollewald, 2022-02-08:
Since I have not heard anything I close this issue now.

closed 2022-02-08 by wollewald

vladkozlov69, 2022-09-28:
Just wanted to put my 5 cents. Today I received from Ali some modules advertised as MPU9250. The magnetometer data is just 0. Acc/Gyr data seems okay. So I believe Ali sellers just got batches of MPU9250 with broken magnetometer and resells them.

wollewald, 2022-09-28:
Hi @vladkozlov69 , thanks for this info. Can you run the example sketch who_am_i.ino ? Maybe you have one of the modules which have the MPU6500 (or other) instead of the MPU9250.

vladkozlov69, 2022-09-28:
Hi @wollewald here it is

`WhoAmI Register: 0x75
Your device is probably an MPU6515
Not sure if it works with this library, just try`