wollewald/MPU9250_WE #2: Conversion error in FIFO readout

savejeff, 2021-08-10:
There is a bug in the readMPU9250xyzValFromFifo function. a the conversion from uin16_t to float does not work with float.
this code is working:
```
Vec3f MPU9250_WE::readMPU9250xyzValFromFifo(){
    uint8_t MSByte = 0, LSByte = 0;
    xyzFloat xyzResult = {0.0, 0.0, 0.0};
    MSByte = readMPU9250Register8(MPU9250_FIFO_R_W);
    LSByte = readMPU9250Register8(MPU9250_FIFO_R_W);
    xyzResult.x = ((int16_t) ((MSByte<<8) + LSByte) * 1.0);
    MSByte = readMPU9250Register8(MPU9250_FIFO_R_W);
    LSByte = readMPU9250Register8(MPU9250_FIFO_R_W);
    xyzResult.y = ((int16_t) ((MSByte<<8) + LSByte) * 1.0);
    MSByte = readMPU9250Register8(MPU9250_FIFO_R_W);
    LSByte = readMPU9250Register8(MPU9250_FIFO_R_W);
    xyzResult.z = ((int16_t) ((MSByte<<8) + LSByte) * 1.0);
    return xyzResult; 
}
```

wollewald, 2021-08-10:
@savejeff , I have just tried it and I obtain also negative values. So, somehow it works as is. And the compiler does not complain. MSByte and LSByte are read as uint8_t, that's right. But 
``xyzResult.x = (MSByte<<8) + LSByte;``
in conjunction with 
`` xyzFloat xyzResult = {0.0, 0.0, 0.0};``
works. The implicite change of varriable type seems to work. 

At least it works with my compiler and I have to admit it's not best coding. Have you tried it and you don't get negative values or is it just an observation you made when you read the code? Just my interest. 

I will do the quick fix now, but I think the brackets should be different: 
You proposed:
``xyzResult.x = ((int16_t) ((MSByte<<8) + LSByte) * 1.0);``

But I think cleaner is:
``xyzResult.x = ((int16_t) ((MSByte<<8) + LSByte)) * 1.0;``

The recommendation to improve the speed by reading all bytes in one go is also good, but I will do this at a later stage.

savejeff, 2021-08-10:
Moin Moin,

i run the code on the new Raspberry Pi Pico.
i think this is caused by different compilers. If you don't specify the explicit conversion from unit to int the behaviour is undefined. 
some compiler thus just convert unsigned in the corresponding float value. 

i only get negative numbers with my proposed code. without I got high postitiv numbers

wollewald, 2021-08-10:
Nice to hear a "moin moin" since I grew up in northern Germany. 

I have changed it now and do the same in the ICM20948 lib.

savejeff, 2021-08-10:
I'm from Germany (new Frankfurt) and I saw the link to the Germany website ;]

closed 2021-08-10 by savejeff