wollewald/MPU9250_WE #3: Feature: tilt adjusted compass output

tmcadam, 2022-01-27:
Thank you for this great library. It is the only one I have found for the mpu9250 that outputs pitch and roll correctly with my sensor. 

I see you have magnetometer x,y,z output. I was wondering if a function can be added to output a tilt adjusted compass heading from these values?

Thanks

ivicajan, 2022-02-08:
It should be easy, based on available info you should add something like this after you read pitch, roll and mag values.
You'll need to define Declination for your location, snip of the code based on the available example at this github:
.....
  pitch = myMPU9250.getPitch();
  roll  = myMPU9250.getRoll();
  
  float Xh = magValue.x*cos(pitch/180*Pi) + magValue.y*sin(roll/180*Pi)*sin(pitch/180*Pi) - \
  			 magValue.z*cos(roll/180*Pi)*sin(pitch/180*Pi);
  float Yh = magValue.y*cos(roll/180*Pi) + magValue.z*sin(roll/180*Pi);
  Xh = Xh * 0.9 + magValue.x * 0.1;
  Yh = Yh * 0.9 + magValue.y * 0.1;
  float Heading = atan2(Xh, Yh) * 180 / Pi;
  Heading += Declination ; 
  
  if (Heading > 360.0) Heading -= 360.0;
  if (Heading < 0.0)   Heading += 360.0;
  // ----- Allow for under/overflow
  if (Heading < 0) Heading += 360 ;
  if (Heading >= 360) Heading -= 360 ;

  Serial.println(Heading);

wollewald, 2022-07-28:
@tmcadam , @ivicajan 
I am closing this issue now. At least for the moment I do not have the capacity to implement this. If it's so easy as ivicajan says, the  I am happy if someone wants to become a contributor!

closed 2022-07-28 by wollewald

IgnacioRubioScola, 2023-10-26:
Hi!

Thank you for this wonderful library!

I think that it would be really nice to have DMP functionalities for the 6500 (and also the 9250).

I found that the DMP has already been implemented for the MPU9250 in:
[SparkFun_MPU-9250-DMP_Arduino_Library](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/tree/master)

Headers and functions for using the DMP are defined in:
[SparkFunMPU9250-DMP.h](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/SparkFunMPU9250-DMP.h
)
[SparkFunMPU9250-DMP.cpp](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/SparkFunMPU9250-DMP.cpp)

Thank you again for your time!