wollewald/MPU9250_WE #5: Compilation issue

RamkumarGorre, 2022-06-29:
Hi, I am new to the Arduino Uno, observed following compilation issues. I tried to change some of the header files to c notation, but it is never ending. There may be an easy way out. Could you help me in fixing this compilation error?
C:\Users\ramgk\OneDrive\Documents\Arduino\libraries\Bolder_Flight_Systems_Eigen\src/Eigen/Core:50:10: fatal error: complex: No such file or directory
 #include <complex>
          ^~~~~~~~~
compilation terminated.

wollewald, 2022-06-29:
Hi @RamkumarGorre , this is not sufficient information to help you. You wrote you try to change header files to c notation. 

So, first question: do you get compilation errors with the original library and original example sketches? 

Second: if you should only get  compilation errors after you have changed something, what have you changed?

And then one hint: your error message suggests that the issue is related with a Bolderflight library and not with my library.

RamkumarGorre, 2022-06-29:
> 

Hi Wolfgang Ewald, Thanks for the quick reply, I just have removed the library and reinstalled. The compilation went through fine now. I am not getting any output on the serial port. are there any check points to see if the MPU9250 is working as expected?

wollewald, 2022-06-29:
In the exmple sketches I have the check:
````
if(!myMPU9250.init()){
    Serial.println("MPU9250 does not respond");
  }
  else{
    Serial.println("MPU9250 is connected");
  }
  if(!myMPU9250.initMagnetometer()){
    Serial.println("Magnetometer does not respond");`
````
You could change the code let an LED blink if all is OK. You can also check if the connection is OK with an I2C scanner sketch, which you find here, for example:
https://wolles-elektronikkiste.de/en/i2c-scanner

Have you set the Baud rate correctly? I am using 115200 in the examples.

RamkumarGorre, 2022-06-29:
> In the exmple sketches I have the check:
>
> ```
> if(!myMPU9250.init()){
>     Serial.println("MPU9250 does not respond");
>   }
>   else{
>     Serial.println("MPU9250 is connected");
>   } [...]

I have added an additional debug statement, the following function call is hung
"myMPU9250.init()" in the if statement. I assume, there is something wrong with the board that i connected. I have connected to VCC - 5V on Arduino
GND - GND on Arduino
SCL - A5 on Arduino
SDA - A4 on Arduino
ADO - GND on Arduino

RamkumarGorre, 2022-06-30:
> 

Hi Wolfgang Ewald, I am able to run the first sketch of yours successfully. Thanks for the code and support. There was a faulty connector, which i had to replace to make it work.

wollewald, 2022-06-30:
I amm happy that it worked. Have fun and stay healthy!

closed 2022-06-30 by wollewald