wollewald/MPU9250_WE #6: Offset function - no sensitivity

RamkumarGorre, 2022-07-02:
Hi Wolfgang Ewald,

I am using the sketch MPU9250_acceleration_data.ino to get the acceleration data. in which there is a function call **myMPU9250.autoOffsets()**, which is meant to use it for the calibration based on the comments in the code. When I used this sketch as-is, I do not see any change in output values despite the movement in the MPU9250 using my hand. But, when commented the **autoOffsets** function call, output values are getting changed with the MPU9250 unit movement with the hand. I am pasting the sample snapshot values after commenting the **autoOffsets** function call.
Raw acceleration values (x,y,z):
1280.00   -16127.00   24577.00
Corrected ('calibrated') acceleration values (x,y,z):
1280.00   -16127.00   24577.00
g values (x,y,z):
0.08   -0.98   1.50
Resultant g: 1.80

I am also pasting the code below for the autoOffsets function call, if you can throw some light on this code function, that may help me to understand and use it.

_void MPU6500_WE::autoOffsets(){
    enableGyrDLPF();
    setGyrDLPF(MPU9250_DLPF_6);  // lowest noise
    setGyrRange(MPU9250_GYRO_RANGE_250); // highest resolution
    setAccRange(MPU9250_ACC_RANGE_2G);
    enableAccDLPF(true);
    setAccDLPF(MPU9250_DLPF_6);
    delay(100);

    xyzFloat accelerationOffsetAccumulator{0.f, 0.f, 0.f};
    xyzFloat gyroOffsetAccumulator{0.f, 0.f, 0.f};
    for(int i=0; i<50; i++){
        // acceleration
        accelerationOffsetAccumulator += getAccRawValues();
        // gyro
        gyroOffsetAccumulator += getGyrRawValues();
        delay(1);
    }

    // acceleration
    accelerationOffsetAccumulator /= 50.f;
    accelerationOffsetAccumulator.z -= 16384.0f;
    accOffsetVal = accelerationOffsetAccumulator;
    // gyro
    gyrOffsetVal = gyroOffsetAccumulator / 50.f;

}_

wollewald, 2022-07-02:
Hi @RamkumarGorre ,

strange. On my side it works. To ensure that nothing messed up in an update or so I just tried it. 

This is what I get:
Raw acceleration values (x,y,z):
1937.00   -899.00   11947.00
Corrected ('calibrated') acceleration values (x,y,z):
10.74   1.34   16386.84
g values (x,y,z):
0.00   0.00   1.00
Resultant g: 1.00

What is also strange is you raw z-value. In theory it should be 16384 (= 1g). The reality differs because the x,y,z-axes will never be exactly 0°,0°,90° as you might expect when you position the MPU9250 horizontally in its x-y-plane. But 24577 is something I can't explain as long as the MPU9250 is not moved during calibration (no movement, no vibration, etc). 

Did you change anything in the sketch or the lib?

And have you read the tutorial with regards to this point and considered all restrictions?

<em>If the MPU9250 is positioned flat, i.e. horizontal x,y-plane, only the acceleration due to gravity acts on it. Accordingly, the g-values for the x- and y-axis should be zero and for the z-axis it should be one. However, these values are more or less shifted. The function autoOffset() measures the offset values which are subtracted from future measured values. The slope, on the other hand, had not to be corrected for the modules I tried out. In the 2g range, the difference between the minimum and maximum raw acceleration values was quite close to the expected 2 x 16384 (i.e. 2x 10^15). autoOffset() only works reliably if:

* the module is positioned flat with its x,y-plane,
* is not moved, and
* the function is called as the first function in the setup (because it changes some settings)
Alternatively, you can use the setAccOffsets() function. It results in less good zero values, but is more accurate with larger angles, and you don’t have to position the module when the program starts.</em>

This should also explain what it does. Then maybe also the code of the function is clearer. It measures the acceleration and gyroscope values 50 times and calculations the mean value. In future measurements these values are subtracted. A very simple method.

RamkumarGorre, 2022-07-03:
> 

Hi Wolfgang Evald,

I have had commented this line as I said
**myMPU9250.autoOffsets()** and now I run the code as-is, following is the output.
Raw acceleration values (x,y,z):
1573.00   1572.00   1571.00
Corrected ('calibrated') acceleration values (x,y,z):
0.00   0.00   16384.00
g values (x,y,z):
0.00   0.00   1.00
Resultant g: 1.00
The above values are not changing with the movement or positioning of the MPU9250.

is there any procedure to find if the h/w unit functioning status?

wollewald, 2022-07-03:
When I comment myMPU9250.autoOffsets(); I also get what I expect: 

Raw acceleration values (x,y,z):
-344.00   -494.00   15731.00
Corrected ('calibrated') acceleration values (x,y,z):
-346.00   -494.00   15735.00
g values (x,y,z):
-0.02   -0.03   0.96
Resultant g: 0.96

So at least on my end it works with and without this function. I have absolutely no explanation why it doesn't work on your side. 

Are you using the _unchanged_ acceleration example sketch (apart from the commented autoOffset() )? And have you applied _any_ changes to the library?

Which microcontroller board do you use? It shouldn't change anything, but I am trying to find out what the difference is. Without being able to reproduce the issue it will be difficult to find a solution.

wollewald, 2022-07-03:
Another question: what are your results if you leave the autoOffset() uncommented?

wollewald, 2022-07-03:
And last question for the moment: what do you mean with h/w unit?

wollewald, 2022-07-03:
Sorry it was not the last question. :-) 
Can you try another library for the MPU9250. If it also doen't work then there is probably an issue with the module or with the circuit. And if another library works then it's something specific with my library.

RamkumarGorre, 2022-07-03:
> Sorry it was not the last question. :-) Can you try another library for the MPU9250. If it also doen't work then there is probably an issue with the module or with the circuit. And if another library works then it's something specific with my library.

When I am referring the hw unit, it is MPU9250. I have not changed anything, using the sketch as-is. is there any procedure or method to find the faults in MPU9250?

RamkumarGorre, 2022-07-03:
Hi Wolfgang Ewald,
I have tried the function call "whoAmI", for which i am getting a response of '0'.  I believe, there must be something is wrong with the board.
WHO AM I0
Raw acceleration values (x,y,z):
1573.00   1572.00   1571.00
Corrected ('calibrated') acceleration values (x,y,z):
0.00   0.00   16384.00
g values (x,y,z):
0.00   0.00   1.00
Resultant g: 1.00

Let me know if there is any other check that i can make.

wollewald, 2022-07-03:
If you are using I2C you could run a scanner to see if the module answers at all and if the I2C address is correct. You can a scanner here:
https://wolles-elektronikkiste.de/en/i2c-scanner

It's not a good sign that "Who am I" returns zero.

I don't have much hope that it helps but you could also try and eleminate everything in the example sketch but the query of the raw values. Does it change anything?

And then you could check if another library works. I have just tried this one: 
https://github.com/asukiaaa/MPU9250_asukiaaa
You can install it via the Arduino IDE directly. The GetData example sketch works fine with my module. 

Good luck. I know very well how exhausting and disappointing this can be!

RamkumarGorre, 2022-07-04:
> 

Hi Wolfgang Ewald,

Thanks for the quick support. I2C scanner example is working, returning the following.

****************************

Scanning in standard mode (100 kHz):
0x69

Scanning in fast mode (400 kHz):
0x69

****************************

Scanning in standard mode (100 kHz):
0x69

Scanning in fast mode (400 kHz):
0x69

****************************

I have tried the MPU9250_asukiaaa library, but getting the following response.

Cannot read sensorId 2
Cannod read accel values 2
Cannot read gyro values 2
Cannot read mag values 2
at 87939ms

Cannot read sensorId 2
Cannod read accel values 2
Cannot read gyro values 2
Cannot read mag values 2
at 88443ms

It looks like, MPU9250 board is having a problem.

wollewald, 2022-07-04:
Do you have an ADO pin on your module? Most modules have it. When you connect it to GND the address should be 0x68 and when connected to HIGH it should be 0x69. My sketches have 0x68 as default. So either you change the I2C address of the board by connecting AD0 to GND (you can see the effect with the I2C scanner) **or** you just change the I2C address in the sketch from:
````
#define MPU9250_ADDR 0x68
````
to
````
#define MPU9250_ADDR 0x69
````
Maybe this solves the problem.

RamkumarGorre, 2022-07-04:
> 

I have changed to 0x69, there is a change in values. My serial output.

WHO AM I112
Raw acceleration values (x,y,z):
-1876.00   288.00   15535.00
Corrected ('calibrated') acceleration values (x,y,z):
-9852.08   -255.08   18607.14
g values (x,y,z):
-0.60   -0.02   1.14
Resultant g: 1.29

I assume, it is working. However, there is a difference in value for WHOAMI. It suppose to be 0x71, but it is coming as 112(0x70).

wollewald, 2022-07-04:
Great that it works. If WHOAMI is 0x70 instead of 0x71 then it seems you have an MPU6500 instead of a MPU9250. Can you run the MPU6500_all_data example sketch? Of course you also need to change to the address 0x69. 
Does it say at the beginning "MPU6500 is connected"?
The difference between the MPU6500 and the MPU9250 is just that the MPU6500 does not have a magnetometer.

RamkumarGorre, 2022-07-04:
This is the starting message, it looks to be MPU9250. 
MPU9250 does not respond
Position you MPU9250 flat and don't move it - calibrating...
Done!
Angle x = 0.01  |  Angle y = 0.04  |  Angle z = 89.80
Pitch   = 0.01  |  Roll    = 0.01

Angle x = 0.03  |  Angle y = 0.06  |  Angle z = 88.57
Pitch   = 0.02  |  Roll    = 0.03

Angle x = 0.00  |  Angle y = 0.02  |  Angle z = 87.46
Pitch   = 0.00  |  Roll    = 0.01

Let me try the MPU6500 all data example sketch

RamkumarGorre, 2022-07-04:
> Great that it works. If WHOAMI is 0x70 instead of 0x71 then it seems you have an MPU6500 instead of a MPU9250. Can you run the MPU6500_all_data example sketch? Of course you also need to change to [...]

Serial output of the MPU6500 all data sketch.

MPU6500 is connected
Position you MPU6500 flat and don't move it - calibrating...
Done!
Acceleration in g (x,y,z):
-0.00   0.00   1.00
Resultant g: 1.00
Gyroscope data in degrees/s: 
-0.07   0.01   0.02
Temperature in °C: 29.12
********************************************
Acceleration in g (x,y,z):
0.00   0.00   1.00
Resultant g: 1.00
Gyroscope data in degrees/s: 
-0.06   -0.00   0.03
Temperature in °C: 29.13
********************************************

How do we know if the module is MPU6500 or MPU9250?

wollewald, 2022-07-04:
The WHO_AM_I is one indicator. In addtion you can try to run a magnetometer sketch and if you have a MPU6500 you will see the message "Magnetometer does not respond". And finally, look on the square sized IC with the 24 pins on the board. The label on it is very tiny. If there is a "MP92" it should be a MPU9250, the MPU6500 should have a label "MP65".

RamkumarGorre, 2022-07-05:
> The WHO_AM_I is one indicator. In addtion you can try to run a magnetometer sketch and if you have a MPU6500 you will see the message "Magnetometer does not respond". And finally, look on the square [...]

Attaching the image of the module, it has the label for MP9250/6500. 
Following is the serial output for the Magneto Meter sketch.
MPU9250 does not respond
Magnetometer does not respond
Magnetometer Data in µTesla: 
0.00   0.00   0.00
Magnetometer Data in µTesla: 
0.00   0.00   0.00
Magnetometer Data in µTesla: 
0.00   0.00   0.00
Magnetometer Data in µTesla: 
0.00   0.00   0.00
[image]

wollewald, 2022-07-05:
I don't mean the bigger white label "MPU-9250/6500". This does not tell anything. It seems the producers use the same basic board and put either MPU9250 ICs or MPU6500 ICs on it. I am talking about the small IC in the middle with the 4x6 pins. It's hard to read what is written on it. You may need a magnifying glass. On this IC you should find either "MP92" or "MP65". On the photo there is no chance to read the IC label.

wollewald, 2022-07-05:
And the output of the magnetometer sketch indicates that it's am MPU6500.

RamkumarGorre, 2022-07-06:
> 

I hardly see any letters on the chip, but the details are printed only on the PCB. The print on PCB, indicates MPU9256, but actually it is not after running multiple tests as suggested by you. I got another board, it looks to be MPU9256 as WHOAMI is returning the correct value of 0x71. I am looking for some documentation on the roll & pitch measurements for your sketch. I want to understand the formula used to compute the roll & pitch and also some description of the function calls used in the sketch. I am also going through the code, but your few words on these items would help me to understand the logic quicker.

wollewald, 2022-07-06:
How simple angles are calculated is something I understand, but roll and pitch calculations exceed my math skills. At the end I have copied the equation. If you look at this article you get an impression how this is calculated, although the equations are slightly different at the end: 
https://mwrona.com/posts/accel-roll-pitch/
If you google "pitch and roll angle from accelerometer" you get lots of additional links.

wollewald, 2022-07-07:
I have just decided to change the pitch calculation to a version that is used in most articles about this topic, like here:

https://wiki.dfrobot.com/How_to_Use_a_Three-Axis_Accelerometer_for_Tilt_Sensing

So, it had a useful side-effect tha you have raised the issue!

wollewald, 2022-07-09:
I think the original issue is solved and I can close.

closed 2022-07-09 by wollewald

RamkumarGorre, 2022-07-12:
> I think the original issue is solved and I can close.

You can close this. Thank you very much for your support. Let me go through the new proposed pitch calculation.

wollewald, 2022-07-12:
Yes I know there is a difference. And the sign of the pitch changes. But I really wanted to take what people usually take.