wollewald/MPU9250_WE #7: Does this library make use of the DMP?

kireita, 2022-07-08:
i have several MPU6500 and i want to get the Yaw value as well, as i see your library only gives pitch and roll and due to the missing magnetometer yaw is dificoult to get (or so i understand).

i am following a [video ](https://www.youtube.com/watch?v=k5i-vE5rZR0) that is about the MPU6050 and they specify i need a library that reads directly from the DMP. hence my question.

wollewald, 2022-07-08:
The answer is very simply no. I tried to implement DMP for few hours, got stuck at a certain point and decided to not pursue this. At least not for the moment... but the moment could be long!

kireita, 2022-07-08:
Thank you for your fast awnser.

Do you have any recomendation for me to get pitch,yaw,roll or quaternions of the mpu6500?

If you need inspiration on how to use DMP this [library](https://github.com/electroniccats/mpu6050) has it implemented. But for the MPU6050, although it is not very different from the MPU9250.

Thank you for your time.

wollewald, 2022-07-08:
I have implemented calculations for pitch and roll. I took the equations from here: 

https://wiki.dfrobot.com/How_to_Use_a_Three-Axis_Accelerometer_for_Tilt_Sensing

If you include gyroscope values then you can calculate yaw. But it is relative to the starting orientation and there's always a certain error in calculation. And since you do not have a reference which the magnetometer could provide, the errors will add up and yaw will drift. 

The easiest approach is to constantly monitor the gyroscop value for the z-Axis (gyroZ). The value is provided in degrees per second so you just multiply it with the time and get yaw this way:

Delta yaw = Delta gyroZ * Delta time.
yaw = yaw + Delta yaw

The higher the frequency you apply (= the smaller Delta time) to measure gyroZ the lower the error.  

And thanks for the link. Maybe I will try it again - but won't promise!

wollewald, 2022-07-28:
I think I have answered the main question of the issue. If I will implement DMP depends on the time I will have.

closed 2022-07-28 by wollewald

IgnacioRubioScola, 2023-10-26:
Hi!

Thank you for this wonderful library!

I think that it would be really nice to have DMP functionalities for the 6500 (and also the 9250).

I found that the DMP has already been implemented for the MPU9250 in:
[SparkFun_MPU-9250-DMP_Arduino_Library](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/tree/master)

Headers and functions for using the DMP are defined in:
[SparkFunMPU9250-DMP.h](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/SparkFunMPU9250-DMP.h
)
[SparkFunMPU9250-DMP.cpp](https://github.com/sparkfun/SparkFun_MPU-9250-DMP_Arduino_Library/blob/master/src/SparkFunMPU9250-DMP.cpp)

Thank you again for your time!