wollewald/MPU9250_WE #8: Sequence of execution of functions in setup arduino sketch

RamkumarGorre, 2022-07-13:
Hi Wolfgang Ewald,

In the setup function call of [MPU9250_acceleration_data.ino] sketch, there is a comment before the setSampleRateDivider function to call this function only if the DLPF filter is enabled. But, the DLPF enable function call is made later. What is the sequence to  follow? I altered the function calls, but do not see any difference in output values.

wollewald, 2022-07-13:
Hi,
the sequence doesn't matter. And at least I see differences if I change the DLPF. You need to look at the raw values. Here I have taken the unchanged settings, i.e. DLPF = 6. The module was just lying unmoved on the table. To see more values to compare I only printed the corrected raw values.
[image]

And here I have changed the DLPF to 0 and you will get similar results if you deactivate the DLPF with  enableAccDLPF(false). Just commenting out is not enough as documented in the tutorial and the function list. 
[image]

I would say this is quite a difference in noise. 

To check if the sample rate divider works I have set the sample rate divider to 255. That means the output rate is 4 Hz. In order to see the effect you need to have an output rate on the serial monitor which is higher. I have applied reduced the delay of 100 ms in the example sketch and as expected I saw that only every second to third value is a new one:
 
[image]

I also applied a sample rate divider of 1 and as expected after 100 ms I always got a new value.

[image]

So, DLPF and a sample rate divider are working. 

The interesting thing I found is that the the sample rate divider seems to work even if I deactivate the DLPF or set it level 0 or 7.

[image]

But since I don't know if there are maybe different generations of the MPU9250 which behave diffenrently, I keep the comment.

wollewald, 2022-07-28:
@RamkumarGorre , can I close this issue now? Or are you still checking?

RamkumarGorre, 2022-07-28:
Hi Wolfgang Ewald,

You can close this case. Thanks for your support.

regards,
Ram Kumar Gorre

-- 
The information contained in this electronic communication is intended 
solely for the individual(s) or entity to which it is addressed. It may 
contain proprietary, confidential and/or legally privileged information. 
Any review, retransmission, dissemination, printing, copying or other use 
of, or taking any action in reliance on the contents of this information by 
person(s) or entities other than the intended recipient is strictly 
prohibited and may be unlawful. If you have received this communication in 
error, please notify us by responding to this email or telephone and 
immediately and permanently delete all copies of this message and any 
attachments from your system(s). The contents of this message do not 
necessarily represent the views or policies of BITS Pilani.

wollewald, 2022-07-28:
You are welcome!

closed 2022-07-28 by wollewald