wollewald/MPU9250_WE #11: Magnetometer as Compass

dyk74, 2022-08-29:
I am using the libraries for a project with arduino; the magnetometer works however it gives me values that are not accurate. I would like to know if it is possible to calibrate the magnetometer (without using the 8 movement) to make it more accurate so that I can then use it as a compass.

wollewald, 2022-08-29:
Hi @dyk74 , it would be great to have a simple one-off magnetometer calibration but I don't think it's possible. One reason is that the magnetic field changes depending on the location, e.g. 
indoor/outdoor, buildings around, metallic parts, etc. I thought a bit and tried a bit when I wrote the library but then I noticed all ideas I had were more complex than I thought when it came to the details. 
Usually people who want to know the absolute orientation of theq

wollewald, 2022-08-29:
... sorry, writing from smartphone and hit the wrong button....
I wanted to write: people use quaternions, but this is also quite complex.
If anyone has a smart idea I am happy to come back to this. For the moment I have not planned to proceed.

closed 2022-08-29 by wollewald

froussel, 2022-11-28:
Hello,

There are those who like to tweak and therefore for whom the more details there are, the more they like.
And then there are those who seek only the useful for them....
Having a compensated compass is an example.
So a calibration process is needed when the assembly is stationary, then after keeping these parameters on condition of always placing the assembly in the same vicinity.
And then when it's on it gives the direction of the magnetic north pole;
Load is up to the user to manage a screen the deviation according to the place etc...

Is it possible to have an example with only an output in ° of the direction from 0° to 360°

wollewald, 2022-11-28:
Hi @froussel, in theory it's easy. Just turn the magnetometer in all directions and determine the max and min values for the x and y axis. You can define e.g. xmax as north xmin as south and use y values to eliminate the ambiguity of values in between (there are always two angles for each axis which will give the same magnetometer value). If you want to make it more robust, you also need to include the z-axis magnetometer value because if the sensor is not held horizontally it influences the x- and y values. 
I have tried this bit but did not achieve satisfying results and then lost motivation spend more time. If anyone finds a reproducible, procedure I am happy to include it. As I wrote before, I currently do not plan to develop something myself.