wollewald/MPU9250_WE #14: Magnetometer does not respond

Wbiu, 2023-01-11:
Hello,

I´ve a bit of an issue getting the Magnetometer up and running, but every else is  working fine.
I´ve 2 equal  MPU9250 ( those blue one´s ), one of them is working fine, but the secound one seems like the Magnetometer is not responding...
After hours of googling and trying, but of course nothing is working and now I don´t know what to do...
I have tried setting the pass through i2c but 0x37 -> 0x02 , but it wont work, unfortunately.
Any idea what I  cant try next?

Thank you in advance.

I changed the code a bit for testing purposes.

Main:

```
#include <Arduino.h>
#include <MPU9250_WE.h>
#include <Wire.h>
#define MPU9250_ADDR 0x68

MPU9250_WE myMPU9250 = MPU9250_WE(MPU9250_ADDR);

void setup()
{
  Serial.begin(115200);
  Wire.begin();
  delay(2000);
  if (!myMPU9250.init())
  {
    Serial.println("MPU9250 does not respond");
  }
  else
  {
    Serial.println("MPU9250 is connected");
  }

  myMPU9250.enableGyrDLPF();
  myMPU9250.setGyrDLPF(MPU9250_DLPF_6); // lowest noise
  myMPU9250.setGyrRange(MPU9250_GYRO_RANGE_250);
  myMPU9250.setAccRange(MPU9250_ACC_RANGE_2G);
  myMPU9250.enableAccDLPF(true);
  myMPU9250.setAccDLPF(MPU9250_DLPF_6); // lowest noise

  if (!myMPU9250.initMagnetometer())
  {
    Serial.println("Magnetometer does not respond T-T ");
  }
  else
  {
    Serial.println("Magnetometer is connected");
  }
  delay(1000);
}

void loop()
{

  /*
    xyzFloat accRaw;
  xyzFloat gyrRaw;
  xyzFloat mgrRaw;
  accRaw = myMPU9250.getAccRawValues();
  gyrRaw = myMPU9250.getGyrRawValues();
  mgrRaw = myMPU9250.getMagValues();
  Serial.print(accRaw.x);
  Serial.print(";");
  Serial.print(accRaw.y);
  Serial.print(";");
  Serial.print(accRaw.z);
  Serial.print(";");
  Serial.print(gyrRaw.x);
  Serial.print(";");
  Serial.print(gyrRaw.y);
  Serial.print(";");
  Serial.print(gyrRaw.z);
  Serial.print(";");
  Serial.print(mgrRaw.x);
  Serial.print(";");
  Serial.print(mgrRaw.y);
  Serial.print(";");
  Serial.println(mgrRaw.z);

  */
}
```
I added some prints for the returning values for debuging in the MPU9250_WE.cpp

```
/************** Magnetometer **************/

bool MPU9250_WE::initMagnetometer(){
    enableI2CMaster();
    resetMagnetometer();

    if (!(whoAmIMag() == MAGNETOMETER_WHO_AM_I_CODE))
    {
        Serial.print("In int_pin ");
        Serial.println(readMPU9250Register8(0x37), HEX);
        Serial.print("I am ");
        Serial.println(whoAmIMag(), HEX);
        return false;
    }
    Serial.print("In int_pin ");
    Serial.println(readMPU9250Register8(0x37), HEX);
    Serial.print("I am ");
    Serial.println(whoAmIMag(), HEX);

    setMagOpMode(AK8963_FUSE_ROM_ACC_MODE);
    delay(10);
    getAsaVals();
    delay(10);
    setMagnetometer16Bit();
    delay(10);
    setMagOpMode(AK8963_CONT_MODE_8HZ);
    delay(10);

    return true;
}
```

Serial out put from the working IMU

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
MPU9250 is connected
In int_pin 2
I am 48
Magnetometer is connected

```
Serial out put from the NOT working IMU

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
MPU9250 is connected 
In int_pin 2
I am 0
Magnetometer does not respond T-T
ets Jul 29 2019 12:21:46

```

wollewald, 2023-01-11:
Hi @Wbiu , even if the modules look the same, they are not necessarily identical. Please read also the reader.md. And please try the MPU9250_who_am_i example sketch.

Wbiu, 2023-01-11:
Hi @wollewald thanks for the fast respond.

The who_am_i skecht says its a MPU9250.
I also added this in to the last section of the setup() function.

```
whoAmICodeMag = myMPU9250.whoAmIMag();
Serial.println(whoAmICodeMag,HEX);      // returns 0
```

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
WhoAmI Register: 0x71
Your device is an MPU9250
0
```

I also took a picture of the imu chip.
The writen text says its a MP92 16... 

[image]

wollewald, 2023-01-11:
Hi @Wbiu , that is really strange. When other people had issues with the magnetometer it was always related with the IC not being an MPU9250. But everything looks OK with your module. And the fact that one module is working and the other doesn't proves that there is no issue with wiring, power supply or anything else. And on my side it's difficult to help you because I can't reproduce the issue. 

What you could do next is to try another library like this:
https://github.com/asukiaaa/MPU9250_asukiaaa 
or this one:
https://github.com/kriswiner/MPU9250

Wbiu, 2023-01-12:
Yes its surtenly strange, but unfortunately both liblary does not provide any Mag readings. 
Since I need Mag readings for the Heading calculations,  I think I need to buy a new one ... .

Well thank you for the quick respondings and help. !

wollewald, 2023-01-12:
At least all show the same result, so clearly an issue with the module. I think it's wise to buy a new module - good luck. I close the issue since it's not related with the library.

closed 2023-01-12 by wollewald