wollewald/MPU9250_WE #15: ESP32C3

MichaelQun, 2023-09-10:
Hello, thank you for developing this library. I've been searching extensively online, and your library is the only one I've been able to get working with the esp32c3.

Unfortunately, this specific part of the code, which you can see here: 
[image]
takes approximately 1 millisecond for each call. As a result, when I need to collect 40 samples it adds up to about 80 milliseconds.
Is there any way to optimize these calls for faster execution? Your assistance would be greatly appreciated!

wollewald, 2023-09-10:
Hi, some comments:
1) are you using I2C or SPI? SPI is much faster. 

2) For both I2C and SPI you can increase the clock speed (setClock() / setSPIClockSpeed();) to a certain extend. I don't know where the limit is for an esp32c3. 

3) Don't use Serial.print() after every getGValuesFromFiFo or getGyrValuesFromFifo or after requesting in pairs. You could store them in an array. Serial.print is quite slow. In your code snippet I can see you measure the time need with the variable f. I assume that you have somewhere in your code something like ``requestTime = millis() - f;`` or so. That should be placed before ``Serial.println("Both commands took: ")`` otherwise you include the time needed for the Serial.println(). 

4) If you have done all that and it is still not fast enough I could in theory an option to read the FIFO data in a burst read and store it  in an array. What holds me back is that this array would counsume 512 Bytes of SRAM. Not a problem for an ESP32 but would be for AVR based Arduino boards or AVR MCUs.

MichaelQun, 2023-09-10:
Thank you for your response. 

I am currently using I2C and i am unable to switch to SPI given my current hardware configuration.

The maximum clock speed is set to 400 kHz, which is the MPU's specification.

I implemented bulk read of a dataset which results in 11 ms read time for all 42 samples.
(fetching one dataset at a time (consisting of 12 bytes))
This speed is good enough for what I need, so I'm going to keep using this setup.

PS: thanks again for the quick response and the whole library!

wollewald, 2023-09-11:
Hi, great that it is OK now. I close this issue.

closed 2023-09-11 by wollewald