wollewald/MPU9250_WE #17: Magnetometer not working, and ho to get yaw?

Podex12, 2023-12-12:
Hello, sorry for maybe asking a stupid question. but i hope someone can help me. I don't know why but my magnetometer value never change it stay on zero all the time. the other function like acce, gyro, roll, pitch works just fine. is the magnetometer not working because the module? i have tried the example only too with no change from myself, but it still the same. also the second question is, how to get yaw value?

wollewald, 2023-12-12:
@Podex12, that's definitely not a stupid question. Many people have this issue.  In most cases the problem is that the MPU9250 module does not have MPU9250  IC , but an MPU6500. Please try the "Who Am I" sketch. What does it tell you? Unfortunately I can't tell you where to get correct modules. I once gave a recommendation but then the shop changed from the good to fake modules.

wollewald, 2024-02-12:
I think there is nothing more to say - I close the issue now.

closed 2024-02-12 by wollewald