wollewald/MPU9250_WE #18: Error when uploading in xiao esp32s3

giannisj5, 2024-02-12:
c:\Users\user\Arduino\libraries\MPU9250_WE\src/MPU6500_WE.h:337:28: error: section attribute not allowed for 'accOffsetVal'
  337 |     RTC_DATA_ATTR xyzFloat accOffsetVal; // to be able to keep offsets after deep sleep
      |                            ^~~~~~~~~~~~
c:\Users\user\Arduino\libraries\MPU9250_WE\src/MPU6500_WE.h:338:28: error: section attribute not allowed for 'gyrOffsetVal'
  338 |     RTC_DATA_ATTR xyzFloat gyrOffsetVal;
      |                            ^~~~~~~~~~~~

wollewald, 2024-02-12:
@giannisj5 Something I tried found its way into the the release 1.2.8. Now it should work again (1.2.9).

giannisj5, 2024-02-13:
thank you very much, very fast fix. Everything is working perfect

closed 2024-02-13 by giannisj5