wollewald/MPU9250_WE #20: Support for MPU6050

fischer-felix, 2025-01-01:
Hi, working on a project and decided to switch from ADXL345 to MPU6050. Since I had already used the ADXL345_WE library before, I would really like to stay within the same “ecosystem” so to speak.

Initially this library worked with the MPU6050, albeit with wrong temperature readings, but that's not important to me.

After flashing again, however, it all of a sudden stopped working, only reporting 0 and 21 °C for the temperature.

I am using an ESP8266, if anyone has any ideas as to why it was working initially but not any more, I would love some help.

wollewald, 2025-01-01:
Hi @fischer-felix , when I read this issue, my initial thought was that the MPU9250 / MPU6050 are too different to cover them both in one lib. But I have just compared the register maps and was surprised to see how similar they are. I will have a deeper look into this and check what it takes to do the adjustments. Before that I am not able to promise any timeline.

wollewald, 2025-01-01:
And not sure why it initially worked.

wollewald, 2025-01-01:
Hi @fischer-felix, it's me already again. I just tried a bit. The init() function will not be completed when the "WHO_AM_I_CODE" does not match the device ID. The device ID of the MPU6050 is 0x68. It's a coincidence that it is the same like the the I2C address.

As a quick fix you can go into MPU6500.h and change line 226 from: 

``static float constexpr WHO_AM_I_CODE                     = 70;``

to 

``static float constexpr WHO_AM_I_CODE                     = 0x68;``

And I have also a theory now why it initially might have worked: the init function wakes up the MPU6500 (6050) with sleep(false). But init() stops before if the ID check fails. You might have woken up the device with another lib or other code before you tried my lib. If you did this without stopping the power-supply in between, the wake status might have "survived". It's the only explanation I have. 

But most important you try the quick and dirty fix I have proposed.

fischer-felix, 2025-01-01:
Thanks for the quick response! 
Changing the WHO_AM_I_CODE did indeed make it work. 
Your hypothesis about why it initially worked is also spot on. I first tried it when the ESP was still attached via USB to the computer from test with other libraries, then went to grab some food and unplugged it. Good to know that I did not only imagine it working...

Just for completeness’s sake, it does still report the temperature about 11°C (according to the thermometer in my room) too low, but I suppose this is easily fixed by adjusting either ROOM_TEMPERATURE_OFFSET or TEMPERATURE_SENSITIVITY (and again, I personally do not need the temperature in my project).

wollewald, 2025-01-03:
Hi @fischer-felix , the correct temperature calculation is easy to fix. I am working on a new release of the library, which will include the MPU6050. Users will then be able to create an MPU6050 object. > 95 % is identical with the MPU6500. Temperature and the wake on motion interrupt are the main differences I have found so far. Checking the functions and making adjustments needs some time.

wollewald, 2025-01-03:
Hi @fischer-felix , I have just launched a new release which supports the MPU6050. But I decided to limit the support to the basic functions. That means you can query accelerometer, gyroscope and temperature values, and you can choose the ranges. You can try this with the MPU6050_all_data.ino example. Advanced functions like WOM Interrupt, FIFO and others do not work properly. The register documentation of the MPU6050 is incomplete, and this makes the work on further adjustments painful and lengthy.

wollewald, 2025-05-01:
Since there was no other comment I close this issue now.

closed 2025-05-01 by wollewald