wollewald/MPU9250_WE #21: Initial Orientation

ctubig, 2025-01-27:
Hi, this is really great library. Is it possible to get the correct orientation in startup, like not requiring mpu9250 to laid flat? For example, my mpu9250 is in 45 degree during the startup it should give values 45 degree when calling getAngles or getRoll. I tried to call getAngles and getRoll in setup function during startup while holding my mpu9250 in 45 degree roll position, but it still returns 0 values.

wollewald, 2025-01-27:
Hi @ctubig , there two options which come in my mind:

1. You do the autocalibration while the module is in flat position (x, y = 0°, z=90°). Then you query the offsets using getAccOffsets(), e.g. ``xyzFloat offsets = myIMU.getAccOffests()``. Write down the values or save them to the EEPROM if this is an option for your microcontroller. In you final sketch, you don't do the autocalibration but use the offset values with ``myIMU.setAccOffsets(offsets)``. 
2. Do the alternative calibration by determining the maximum and minimum raw values for +/-x,y,z as described in MPU9250_calibration.ino. Use these values with the function myIMU.setAccOffsets(xMin,xMax,yMin,yMax,zMin,zMax); You could then measure the angles when the module is placed flat and substract these values from your future measured angles. 

Please note that measuring angles becomes less accurate the more you get in the region of +90 or -90°. This is because I am using the arcsin function: angle = arcsin(g-value). A &Delta;g-value around 0° has a much higher effect on the angle than the same &Delta;g-value around 90°.

ctubig, 2025-01-27:
I gonna try it. Thank you!